sparkfun/Qwiic_Py #1: Line 142 causes qwiic.list_devices() not to work for other devices

n1che, 2019-12-30:
I'm not sure of the purpose line 142 which adds all of the I2C addresses to _AVAILABLE_I2C_ADDRESS.  

It doesn't appear that the I2C address of the VL531L1X can be changed (but I haven't looked exhaustively).  From my limited looking, it appears as if it is fixed at 0x52.

The inclusion of this line of code means that list_devices() will respond to any I2C device with 'Qwiic 4m Distance Sensor (ToF)', 'QwiicVL53L1X'.

I am hesitant to remove it without understanding why it is there.

santaimpersonator, 2019-12-30:
I am transferring this issue to the Qwiic_Py, since this is the repository the issue is in reference to.

santaimpersonator, 2019-12-30:
As mentioned, the method hasn't been fully worked out as it didn't account for devices with address ranges that extent to all available addresses on the I2C bus. The primary purpose of this package is to tie our available I2C packages to an I2C driver package, under a single python package. We will look into updating the method eventually, but as it isn't doesn't affect the primary function of this package, it will not be a priority task. We are focusing our efforts on additional I2C packages.

* I am closing this issue as it digresses away from the primary issue of the `list.devices()` method. I will repost another issue to log that it needs to be looked into.

Specifically, with the VL53L1X, based on our discussion with the manufacturer, you cannot access the EPROM. Secondly, since that device doesn't store its new I2C address, so you'd have to configure each one as it is added to the I2C bus anyways, so you would have a list of the I2C addresses they were on.

* Unfortunately, the inclusion of the VL53L1X "*issue*" is specific to your application and more of a technical support request. That part of your "*issue*" should have been posted in our [forum](https://forum.sparkfun.com/index.php).

In regards to the code modifications, the addresses are called [here](https://github.com/sparkfun/Qwiic_Py/blob/master/qwiic/__init__.py#L276). Additionally, you have the option to just modify the `available_addresses` variable of the I2C packages.

closed 2019-12-30 by santaimpersonator