sparkfun/Qwiic_Py #2: Issue with list.devices()

santaimpersonator, 2019-12-30:
As pointed out by a user (#1), with the addition of I2C packages with a wide range of available addresses, this method becomes "*buggy*".

This method will be looked at, eventually:
* It doesn't affect the primary functionality of this package as a wrapper to tie our I2C packages and I2C driver under a single package.
* The logic for this method is complex and will take a while to find an appropriate solution.
  * Possible modification could be to list all possible devices for the address.

czanik, 2020-12-19:
An ugly workaround:
```
pip3 uninstall sparkfun-ublox-gps
```
But at least the sample code from the README now works:
```
pi@raspberrypi:~ $ python3
Python 3.7.3 (default, Jul 25 2020, 13:03:44) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import qwiic
Error loading module `ublox_gps`: No module named 'ublox_gps'
>>> results = qwiic.list_devices()
>>> print(results)
[(16, 'Qwiic Titan GPS', 'QwiicTitanGps'), (91, 'Qwiic CCS811', 'QwiicCcs811'), (119, 'Qwiic BME280', 'QwiicBme280')]
>>> 
```

santaimpersonator, 2021-01-29:
@czanik I am going to hide your workaround, since it is more relevant to issue #8 and has caused some confusion... see https://github.com/sparkfun/Qwiic_Py/issues/12#issuecomment-770079723
However, I will copy over and note your workaround in issue #8.

santaimpersonator, 2021-01-29:
**Update:** Currently, the `list_devices()` method is inoperable and there is an additional conflict from the Ublox GPS module's lack of listed addresses because it utilizes serial communication (https://github.com/sparkfun/Qwiic_Ublox_Gps_Py/issues/8).

If you are referencing this issue, please refrain from creating unnecessary duplicates.

santaimpersonator, 2021-05-13:
This issue has been resolved with this commit: https://github.com/sparkfun/Qwiic_Py/commit/0b923c93b38bea51907b60ae4ed419f33bd5d2cc
*Make sure that you are utilizing the package version 1.0.23 or later.*

Closing

closed 2021-05-13 by santaimpersonator