sparkfun/Qwiic_Py #3: Support for Qwiic Scale?

arbernat, 2020-02-28:
I'd like to use this for four qwiic scale devices (https://www.sparkfun.com/products/15242, chipset NAU7802). Has there been any initial effort in this area? Anything I can help with?

Billy-Bogardus, 2020-03-01:
You will have to use An I2C multiplexer. The Data Sheet says that the Device address is permanent.
This Product (https://www.sparkfun.com/products/14685) should get you going.

arbernat, 2020-03-01:
That's good to know - I didn't realize the devices had the same i2c address. I haven't been able to get one at a time working; they're detected as a default instead and I didn't see any mention of the chipset in the python source. I'll take a look at the arduino examples and start porting work.

artemtk, 2020-03-26:
I am interested in the Python code for Qwiic Scale as well. Seems like Qwiic_Py has support for most of Qwiic devices except the scale one. We need to use Scale in our project in addition to the other sensors. Let me know if the work has started before I start porting myself the Arduino code to Python. Thank you.

santaimpersonator, 2020-06-01:
One of our customers ported our Arduino library to Python and linked it for other users:

https://www.sparkfun.com/products/15242#comment-5eb077fac35147318950d323

santaimpersonator, 2021-01-29:
**Update:** I am working on a SparkFun package in my spare time.

santaimpersonator, 2024-07-10:
While the `README.md` still needs to be updated, my coworker has created a SparkFun Python package for the Qwiic scale:
https://github.com/sparkfun/Qwiic_NAU7802_Py

closed 2024-07-10 by santaimpersonator