sparkfun/Qwiic_Py #5: Unable to install on linux

joemcmanus, 2020-07-31:
Hey Sparkfuners - I am unable to install this in a few linux boxes that I tried they all fail around the gps driver:
equests.exceptions.HTTPError: 404 Client Error: Not Found for url: https://pypi.org/simple/sparkfun-qwiic-ublox-gps/

pip or pip3 both throw the error, along with setup.py intstall. 

Any help would be appreciated! 

Thanks,
-Joe

twongCMU, 2020-08-01:
I'm hitting this problem as well

santaimpersonator, 2020-08-06:
Sorry about that, we were re-implementing the GPS package (due to the clock stretching issues) and the sub-module got renamed, which broke the installation. (*Refer to the following commit: https://github.com/sparkfun/Qwiic_Py/commit/6fc1369283d59db989518f00248552852b339870*)

(*\*I will be closing this issue as the installation should be working now.)*

closed 2020-08-06 by santaimpersonator