sparkfun/Qwiic_Py #7: Problems running Python sample code on Raspberry Pi

TaylorHokanson, 2020-08-13:
I have a Raspberry Pi connected to a [Qwiic illuminated button](https://learn.sparkfun.com/tutorials/sparkfun-qwiic-button-hookup-guide/introduction) with a [Qwiic Shim](https://learn.sparkfun.com/tutorials/qwiic-shim-for-raspberry-pi-hookup-guide/all). When I run the following sample code from the README:
```
import qwiic
results = qwiic.list_devices()
print(results)
mydevice = qwiic.create_device(results[0][0])
print(mydevice)
```
I get the following errors:
```
Error loading module `qwiic_tca9548a`: invalid syntax (qwiic_tca9548a.py, line 70)
Error loading module `ublox_gps`: invalid syntax (core.py, line 20)
[(111, 'Qwiic 4m Distance Sensor (ToF)', 'QwiicVL53L1X')]
<qwiic_vl53l1x.QwiicVL53L1X object at 0x7661f790>
```
It appears that my button is being interpreted as a distance sensor, along with some other module issues. Could this be something wrong on my end, or does it indicate a problem with the repo?

Thanks! -T

TaylorHokanson, 2020-08-14:
I may have answered my own question. It appears that the Qwiic button is not among the devices supported by this repo, but information on using it outside the Arduino framework can be found [here](https://cdn.sparkfun.com/assets/learn_tutorials/1/1/0/8/Qwiic_Button_I2C_Register_Map.pdf). I got some basic communication going with the following code:
```
import smbus
channel = 1

# address of the target i2c device
address = 0x6F
# thing you want to do, see link above
# LED brightness is 0x19
request = 0x19

bus = smbus.SMBus(channel)

# this is how you write something
bus.write_byte_data(address, request, 255)

# this is how you read something
#numbers = bus.read_byte_data(address, request)
#print(numbers)
```
I'll leave the issue open just in case any of the authors have a better solution to offer.

santaimpersonator, 2020-10-08:
I filed issues for the error messages. However, this issue is partially a duplicate of https://github.com/sparkfun/Qwiic_Py/issues/2 and partially due to the fact that the Qwiic Button doesn't have a Python package associated with it.

I will be closing this issue because the associate problems have been filed with their respective packages.

closed 2020-10-08 by santaimpersonator