sparkfun/Qwiic_Py #9: Feature Request: Qwiic Button

santaimpersonator, 2020-10-08:
In reference to https://github.com/sparkfun/Qwiic_Py/issues/7, the customer was trying to use this package with the Qwiic Button.

tudormaerean, 2020-12-16:
Dear team,

It would be great to have support for the Qwiic button into the python library. At the moment there's devices like the Servo pHat that have python support, but others like the illuminated button are supported on Arduino. This fragments the possible projects.

Is it possible to get an ETA where the Qwiic illuminated button would/will be supported?

santaimpersonator, 2020-12-16:
@tudormaerean We are working on a backlog of Python support for Qwiic devices, when we can. The Qwiic button is already on our radar... hence the issue filed here. Unfortunately, there isn't an established timeline for the backlog.

santaimpersonator, 2021-01-29:
**Update:** We have preliminarily released a package fore the Qwiic Button (https://github.com/sparkfun/Qwiic_Button_Py). Feel free to check it out,  it has already been integrated into the this package (`sparkfun-qwiic`)... see https://github.com/sparkfun/Qwiic_Py/commit/48b63f8ab5a22c15b9d551837dbc64693371e3ef.

closed 2021-01-29 by santaimpersonator