sparkfun/Qwiic_Py #12: Pi400 qwiic phat extension: list_devices: devices identify wrong

nuclearmistake, 2021-01-08:
After fixing https://github.com/sparkfun/Qwiic_Py/issues/2, list_devices is reporting that I have 2 '4m range sensors', but I definitely have a thermometer and a SerLCD connected.

Can "i2cset" characters to the SerLCD, but was hoping qwiic would be quick and painless.

Can grab exact output this evening.

santaimpersonator, 2021-01-29:
Fixing #2? Can you elaborate on your fix?

Otherwise, I have a feeling your issue is still the same as issue #2. In which case, I will reference this issue and close it out.

nuclearmistake, 2021-01-29:
I pip uninstalled the GPS package -- the workaround for #2.

My issue is that the output of list_devices shows 2 atmospheric sensors, when neither attached device is an atmospheric sensors.

santaimpersonator, 2021-01-29:
That really isn't a workaround... it's just remove the module that uses serial communication (as mentioned in issue #8). I am going to hide their comment to reduce future confusion.

Currently, this is the same as issue #2, so I will be closing it. The `list_devices()` method is currently inoperable until we find a better way to make it work. Feel free to subscribe to issue #2

closed 2021-01-29 by santaimpersonator