sparkfun/Qwiic_Py #24: Raspberry Pi 400 Issue

butofcourse, 2024-06-21:
I'm having trouble with the local installation on my Raspberry Pi 400. I am able to successfully complete the direct install and build a package, but I keep getting an error for the package file installation. I have confirmed that the file is in the correct folder. Any suggestions?

Here is the error: 
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing ./sparkfun_qwiic-1.1.10.tar.gz
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-9arnukn4/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-9arnukn4/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-k4plsrk9
         cwd: /tmp/pip-req-build-9arnukn4/
    Complete output (5 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-req-build-9arnukn4/setup.py", line 45, in <module>
        with io.open(path.join(here, "DESCRIPTION.rst"), encoding="utf-8") as f:
    FileNotFoundError: [Errno 2] No such file or directory: '/tmp/pip-req-build-9arnukn4/DESCRIPTION.rst'
    ----------------------------------------
WARNING: Discarding file:///home/oepuser/Qwiic_Py/dist/sparkfun_qwiic-1.1.10.tar.gz. Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.

santaimpersonator, 2024-07-02:
Could you provide more information beyond, just an error message? It would be helpful for our team to know what you did to cause it (i.e. what command/script you ran)

Additionally, why are you trying to build the package?

- As mentioned in the [README.md](https://github.com/sparkfun/Qwiic_Py?tab=readme-ov-file#installation) and [documentation](https://qwiic-py.readthedocs.io/en/latest/index.html#id8), this package can be installed with `pip`. 
- If you are running into [issues with the `pip` installation because you you are utilizing the Bookworm OS](https://www.raspberrypi.com/documentation/computers/os.html#python-on-raspberry-pi), you have a few options:
    - Use an older distribution of Raspberry Pi OS or Raspbian
    - Use the `--break-system-packages` flag to force install the Python package
        ```
        python3 -m pip install sparkfun-qwiic --break-system-packages
        ```

santaimpersonator, 2024-07-09:
Since we haven't heard back, I'm going to close this issue for now. However, feel free to re-open it, when you have more information to provide.

closed 2024-07-09 by santaimpersonator