sparkfun/Qwiic_Py #25: Import Failure. Invalid driver class name. Unable to locate QwiicAs6212/QwiicKx13X

Aser-Abdelfatah, 2024-07-08:
Hi,
I'm using Nvidia Jetbot with JetPack 4.6. I'm using the standard Spark Fun setup and double checked that the wiring are right according to SparkFun assembly guide [here](https://learn.sparkfun.com/tutorials/assembly-guide-for-sparkfun-jetbot-ai-kit-v20/introduction) 
I installed this package using pip3 as well as Qwiic_I2C and all the drivers.
When I try to run 
`import qwiic`
I get the errors
`Invalid driver class name. Unable to locate QwiicAs6212 `
` Invalid driver class name. Unable to locate QwiicKx13X`
The packages for both ` QwiicAs6212 ` and ` QwiicKx13X ` are installed. Also, the jetbot is not wired to either of the two modules as well. I don't get why I'm receiving the error.
 Furthermore,  `qwiic.scan()` continues to run forever. So, I'm not able to use the Robot class of jetbot, which depends on qwicc.

sfe-SparkFro, 2024-07-09:
Hi there, thanks for reporting this! I can replicate the problem on my end, will work on fixing this soon!

sfe-SparkFro, 2024-07-09:
Ok, so the root problem is that those 2 packages don't match the format that Qwiic_Py expects in order to include them. Relevant code is here:

https://github.com/sparkfun/Qwiic_Py/blob/50964b60e787bbee51f8ba7f11fdff496a04fd0e/qwiic/__init__.py#L156-L202

In short, it expects the device package names to be in the format of `qwiic_device_name`, and tries to find a class in the format of `QwiicDeviceName`. If so, that class is included. If it's not found, that error message gets printed instead.

The reason the [qwiic_kx13x](https://github.com/sparkfun/Qwiic_KX13X_Py) fails is because it has classes named `QwiicKX13XCore`, `QwiicKX132`, and `QwiicKX134`, but not `QwiicKx13X` like excepted. [qwiic_as6212](https://github.com/sparkfun/Qwiic_AS6212_Py) similarly fails because it has a class named `QwiicAs6212Sensor`, but not `QwiicAs6212 `.

I'm not actually sure how best to address this. `qwiic_as6212` could probably be fixed by adding an empty class called `QwiicAs6212` that inherits `QwiicAs6212Sensor`, which should fix this problem and maintain backwards compatibility. However Qwiic_Py really needs to include `QwiicKX132` and `QwiicKX134` separately, but they're both part of the same package, and Qwiic_Py is really not setup to include multiple classes from the same package. Could maybe add a special case to detect if the driver is called `qwiic_kx13x`, then attempt to include both of those classes explicitly, but that's kinda hacky. Will have to mull this over.

So that's why those error messages appear, but it's odd that this wasn't a problem before. For some reason, previous versions of Qwiic_Py seemed to not include those 2 device drivers. If I run `pip install sparkfun-qwiic==1.1.6` (working version as of 3 years ago) followed by `ls /home/pi/.local/lib/python3.11/site-packages/qwiic/drivers/`, I get the following:

```
pi@raspberrypi:~ $ pip install sparkfun-qwiic==1.1.6
...
Successfully installed sparkfun-qwiic-1.1.10
pi@raspberrypi:~ $ ls /home/pi/.local/lib/python3.11/site-packages/qwiic/drivers/
pi_servo_hat        qwiic_ccs811               qwiic_joystick    qwiic_pir        qwiic_serlcd     qwiic_vl53l1x
qwiic_adxl313       qwiic_dual_encoder_reader  qwiic_keypad      qwiic_proximity  qwiic_sgp40      top_phat_button
qwiic_alphanumeric  qwiic_eeprom               qwiic_max3010x    qwiic_relay      qwiic_tca9548a   ublox_gps
qwiic_bme280        qwiic_gpio                 qwiic_micro_oled  qwiic_rfid       qwiic_titan_gps
qwiic_button        qwiic_icm20948             qwiic_pca9685     qwiic_scmd       qwiic_twist
```

`qwiic_kx13x` and `qwiic_as6212` are not included there for some reason, [despite both of those existing in the repo at that time](https://github.com/sparkfun/Qwiic_Py/tree/8282d0f75684f3942ec3c3ab586f8a5a80710860/qwiic/drivers).

However if I run `pip install sparkfun-qwiic==1.1.10` (latest version of this package, skipping broken versions), they are included:

```
pi@raspberrypi:~ $ pip install sparkfun-qwiic==1.1.10
...
Successfully installed sparkfun-qwiic-1.1.10
pi@raspberrypi:~ $ ls /home/pi/.local/lib/python3.11/site-packages/qwiic/drivers/
pi_servo_hat               qwiic_eeprom      qwiic_max3010x      qwiic_relay                 qwiic_twist
qwiic_adxl313              qwiic_gpio        qwiic_micro_oled    qwiic_rfid                  qwiic_vl53l1x
qwiic_alphanumeric         qwiic_icm20948    qwiic_oled_base     qwiic_scmd                  top_phat_button
qwiic_as6212               qwiic_joystick    qwiic_oled_display  qwiic_serlcd                ublox_gps
qwiic_bme280               qwiic_keypad      qwiic_otos          qwiic_sgp40
qwiic_button               qwiic_kx13x       qwiic_pca9685       qwiic_soil_moisture_sensor
qwiic_ccs811               qwiic_large_oled  qwiic_pir           qwiic_tca9548a
qwiic_dual_encoder_reader  qwiic_led_stick   qwiic_proximity     qwiic_titan_gps
```

Not sure why that is, but they're at least being included now, so I'm not going to dive into that.